chirimen-oh/chirimen-drivers #5: 既存の@chirimen-raspiにあるドライバー群の移行

kou029w, 2019-10-24:
既存の[@chirimen-raspi](https://www.npmjs.com/org/chirimen-raspi)にあるドライバー群を[@chirimen](https://www.npmjs.com/org/chirimen)に移行する

kou029w, 2019-10-24:
命名規則については `@chirimen/{チップの型番}` たとえば `npm i @chirimen/adt7410` くらい短くしても良いと思った
実際 r.chirimen.org はすでに /adt7410.js
https://github.com/chirimen-oh/r.chirimen.org/blob/e69bbde6ea796e0184086a119f094bcaf7b6a26e/_redirects#L87

gurezo, 2019-10-24:
@kou029w 
GPIOとI2C両方使える場合は、どうしますか？
例としてはこんなイメージですけど、どうでしょうか？
`npm i @chirimen/gpio/adt7410`
`npm i @chirimen/i2c/adt7410`

kou029w, 2019-10-24:
> `npm i @chirimen/gpio/adt7410`

おそらく残念ながらこのように/複数含むのはnpmの規約違反じゃないかなと思います

gurezo, 2019-10-24:
そうですね、現時点で自分も思いつかないです。
GPIOとI2Cどちらも使える素子がでるまでは、一旦今の方針でやりましょう。
次回のミーティングの議論しても良いし。
程度割り切りも必要ですし。
自分から持ちかけてといて、すみません。

kou029w, 2019-10-27:
READMEなどまだですがPublishしたのでクローズします

- [@chirimen/ads1015](https://npmjs.org/package/@chirimen/ads1015)
- [@chirimen/ads1x15](https://npmjs.org/package/@chirimen/ads1x15)
- [@chirimen/adt7410](https://npmjs.org/package/@chirimen/adt7410)
- [@chirimen/ak8963](https://npmjs.org/package/@chirimen/ak8963)
- [@chirimen/amg8833](https://npmjs.org/package/@chirimen/amg8833)
- [@chirimen/arduino-stepping-motor](https://npmjs.org/package/@chirimen/arduino-stepping-motor)
- [@chirimen/bh1750](https://npmjs.org/package/@chirimen/bh1750)
- [@chirimen/bme280](https://npmjs.org/package/@chirimen/bme280)
- [@chirimen/bmp180](https://npmjs.org/package/@chirimen/bmp180)
- [@chirimen/bmp280](https://npmjs.org/package/@chirimen/bmp280)
- [@chirimen/canzasi](https://npmjs.org/package/@chirimen/canzasi)
- [@chirimen/gp2y0e03](https://npmjs.org/package/@chirimen/gp2y0e03)
- [@chirimen/grove-accelerometer](https://npmjs.org/package/@chirimen/grove-accelerometer)
- [@chirimen/grove-gesture](https://npmjs.org/package/@chirimen/grove-gesture)
- [@chirimen/grove-light](https://npmjs.org/package/@chirimen/grove-light)
- [@chirimen/grove-oled-display](https://npmjs.org/package/@chirimen/grove-oled-display)
- [@chirimen/grove-touch](https://npmjs.org/package/@chirimen/grove-touch)
- [@chirimen/mpu6050](https://npmjs.org/package/@chirimen/mpu6050)
- [@chirimen/mpu6500](https://npmjs.org/package/@chirimen/mpu6500)
- [@chirimen/neopixel-i2c](https://npmjs.org/package/@chirimen/neopixel-i2c)
- [@chirimen/pca9685](https://npmjs.org/package/@chirimen/pca9685)
- [@chirimen/pca9685-pwm](https://npmjs.org/package/@chirimen/pca9685-pwm)
- [@chirimen/pcf8591](https://npmjs.org/package/@chirimen/pcf8591)
- [@chirimen/s11059](https://npmjs.org/package/@chirimen/s11059)
- [@chirimen/sht30](https://npmjs.org/package/@chirimen/sht30)
- [@chirimen/tcs34725](https://npmjs.org/package/@chirimen/tcs34725)
- [@chirimen/veml6070](https://npmjs.org/package/@chirimen/veml6070)
- [@chirimen/vl53l0x](https://npmjs.org/package/@chirimen/vl53l0x)

https://www.npmjs.com/org/chirimen

closed 2019-10-27 by kou029w