chirimen-oh/chirimen-drivers #11: VL53L0X Node.jsで実行中にOperation errorが出る。

rasp753, 2019-12-04:
vl53l0xのnode.js用example (node-examples/vl53l0x/main.js)を実行中にOperationErrorが発生し、強制終了する。何らかの原因により、センサーの値の読み取りに失敗しエラーが発生する模様。

```
pi@raspberrypi:~/chirimen-drivers/node-examples/vl53l0x $ node main.js 
Start Init
(node:6613) UnhandledPromiseRejectionWarning: OperationError: Error: , Remote I/O error
    at bus.readByte.catch.error (/home/pi/chirimen-drivers/node-examples/vl53l0x/node_modules/node-web-i2c/index.js:49:23)
(node:6613) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:6613) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

kou029w, 2020-02-05:
@rasp753 #8 にて確認した際には問題ありませんでしたか

再現していないようであれば、おそらくモジュール、サンプルコードの問題ではなく、何らかの回路、ハードウェアの一時的問題とし、本件クローズしたいと考えます

kou029w, 2020-02-05:
再現するようであれば、その条件をより詳しく知りたいです

rasp753, 2020-02-06:
再度確認したところ、問題を再現できませんでした。

closed 2020-02-06 by rasp753