chirimen-oh/chirimen-drivers #8: 動作確認しているものに関してはnode.js用サンプルコードを提供

kou029w, 2019-12-04:
Verified official driversのなかで動作確認しているものは以下のモジュール
- [ ] ads1015
- [ ] adt7410
- [ ] gp2y0e03
- [ ] pca9685
- [ ] vl53l0x
CHIRIMEN Examplesのなかのコードに倣って、node.jsからも利用できるようにサンプルコードを提供する

kou029w, 2020-01-30:
アップデートされた現在最新のパッケージ

- [ ] @chirimen/ads1015@1.0.1-alpha.1
- [ ] @chirimen/adt7410@1.0.1-alpha.0
- [ ] @chirimen/gp2y0e03@1.0.1-alpha.0
- [ ] @chirimen/grove-accelerometer@1.0.0-alpha.1
- [ ] @chirimen/grove-gesture@1.0.0-alpha.1
- [ ] @chirimen/grove-light@1.0.0-alpha.4
- [ ] @chirimen/grove-oled-display@1.0.0-alpha.1
- [ ] @chirimen/grove-touch@1.0.0-alpha.1
- [ ] @chirimen/pca9685@1.0.1-alpha.0
- [ ] @chirimen/s11059@1.0.2-alpha.0
- [ ] @chirimen/veml6070@1.0.0-alpha.1
- [ ] @chirimen/vl53l0x@1.0.3-alpha.0

rasp753, 2020-01-30:
ADT7410でテスト時にOperationErrorが発生　#23 
node用サンプルは作成済みのため、エラー解消後再テスト

rasp753, 2020-02-04:
Grove touch sensorのNode.js用exampleでbindings.jsにエラーが発生 #28

kou029w, 2020-02-05:
主要なモジュールは一通り動作確認できたのでクローズ

closed 2020-02-05 by kou029w