chirimen-oh/chirimen-drivers #28: grove touch Node.js用exampleでbindings.jsにエラーが発生する。

rasp753, 2020-01-30:
Grove touch sensor (MPR121) のNode.js版example実行時にbindings.jsにエラーが生じ、実行できない。

```
/home/pi/chirimen-drivers/node-examples/mpr121/node_modules/bindings/bindings.js:121
        throw e;
        ^

Error: /home/pi/chirimen-drivers/node-examples/mpr121/node_modules/i2c-bus/build/Release/i2c.node: file too short
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:805:18)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:690:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at bindings (/home/pi/chirimen-drivers/node-examples/mpr121/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/home/pi/chirimen-drivers/node-examples/mpr121/node_modules/i2c-bus/i2c-bus.js:4:32)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
```

gurezo, 2020-01-30:
issue ありますね
https://github.com/adafruit/node_mpr121/issues/5

kou029w, 2020-02-05:
@gurezo 対象のファイルの行番号とエラーメッセージが異なるので今回の問題とはおそらく無関係じゃないですかね?

kou029w, 2020-02-05:
@rasp753 エラーメッセージ "file too short" とあるようですがこれは /dev/i2c-1 への読み書きの際に発生しているものと想像します
例えば、他に /dev/i2c-1 を利用するプロセスが存在し競合しているとこの問題が発生しうるのではないかと想像するのですが、CHRIMENでブラウザでi2cのサンプルコードを動作させていたりなど、そういったことをしていたかどうか確認ください

kou029w, 2020-02-05:
再現条件をより詳しく知りたいです

rasp753, 2020-02-05:
エラーが出ていたファイル
`/home/pi/chirimen-drivers/node-examples/mpr121/node_modules/i2c-bus/build/Release/i2c.node`
が空ファイルになっていました。
npmで再度パッケージを入れ直したところ空ファイルではなくなっており、exampleも正常に動作しました。

kou029w, 2020-02-05:
なるほど/dev/i2c-1ではなくi2c.nodeが存在しなかったわけですね
インストール時にnode-gypのプロセスで失敗して作成されなかった、という感じかもしれませんが、いずれにせよドライバーの問題ではなかった、ということで本件クローズします。

closed 2020-02-05 by kou029w