chirimen-oh/chirimen-drivers #54: PCA9685-PWM init時にRemote I/O error が発生する

rasp753, 2020-02-13:
pca9685をNode.jsで動かした際にinit()時にRemote I/O errorが発生する。

ブランチpca9685-pwmのnode-example/pca9685-pwm/example1.jsで発生。
配線はschematic1.pngの通りに配線しました。

エラーログ
```
(node:2745) ExperimentalWarning: The fs.promises API is experimental
a
b
(node:2745) UnhandledPromiseRejectionWarning: OperationError: Error: , Remote I/O error
    at Object.write8 (/home/pi/chirimen-drivers/node-examples/pca9685-pwm/node_modules/node-web-i2c/index.js:58:27)
(node:2745) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:2745) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

kou029w, 2020-02-13:
SDカードの中身の途中のコードを見ると

```js
  await pca9685_pwm.init();
```

と書かれておりましたが、いまの@chirimen/pca9685-pwmの実装では引数に周波数を取る実装になってます

実装:
https://github.com/chirimen-oh/chirimen-drivers/blob/241b1e81800861efa24837ca8ed3299584b095d5/packages/pca9685-pwm/pca9685-pwm.mjs#L18

このため、引数を省略した際に参照出来ず何らかの不正な値に暗黙的な変換が行われたのではないかと想像しますが、この点に関していかがでしょうか
ご確認の程よろしくお願いします。

使い方に関してはCHIRIMENのExamplesのサンプルコードを参照:
https://github.com/chirimen-oh/chirimen/blob/fca0d6d5f6a257f68d4406e20f081b868285f3c4/gc/contrib/examples/gpio-i2c-PWMHBridge1/main.js#L18
https://github.com/chirimen-oh/chirimen/blob/fca0d6d5f6a257f68d4406e20f081b868285f3c4/gc/contrib/examples/gpio-i2c-PWMHBridge2/main.js#L13

kou029w, 2020-02-13:
> 引数を省略した際に参照出来ず
と思ったらデフォルト60Hzになるように書かれてました
となると原因は何でしょうかねぇ。。
例えばブラウザのCHIRIMENと同時使うなど他に干渉する要素が無さそうなら、単純にI2CSlaveDevice.write()を繰り返し呼び出すだけでエラーになる、ということですかねぇ
https://github.com/chirimen-oh/chirimen-drivers/blob/241b1e81800861efa24837ca8ed3299584b095d5/packages/pca9685-pwm/pca9685-pwm.mjs#L25-L30