chirimen-oh/chirimen-drivers #89: [MPU6050] 取得される値の異常

kou029w, 2020-10-04:
サンプルコード実行すると温度が高い (25℃くらい→41℃くらいが取得される)

データシート

https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6000-Register-Map1.pdf
P30 4.18 Registers 65 and 66 – Temperature Measurement TEMP_OUT_H and TEMP_OUT_L

本来2バイトをそれぞれ取得する必要があるのではないかと考えるが、実装では次の通り一度に16ビット取得しようとしているように見える

https://github.com/chirimen-oh/chirimen-drivers/blob/2dedfa876c7f8101db742d67e4bb57e44221b944/packages/mpu6050/mpu6050.mjs#L32

kou029w, 2020-10-04:
Arduino 用のライブラリを参照すると一度に16ビットすることができる様子なので謎 https://github.com/jarzebski/Arduino-MPU6050

kou029w, 2020-10-04:
https://github.com/chirimen-oh/chirimen-drivers/blob/fbd1adf649e42bdb924b606b83d1af8fe57a33a5/packages/mpu6050/mpu6050.mjs#L69

ここ `>` が一個多い…?

satakagi, 2020-12-10:
MPU6060だと思って、MPU9250使ってませんか？　そうすると似たような挙動が起きるはず（温度センサーレジスタ番号まで同じだけど物理量算出する定数だけ違う感じ）