chirimen-oh/chirimen-drivers #95: ES Module の書き方 (モジュール以前のブラウザコードから ES Module への移行方法) を説明する

dynamis, 2020-12-10:
現在本リポジトリのドライバは原則として mjs ベースで開発、js (UMD) を自動生成というか達になっているが、[推奨モジュール構成](https://chirimen.org/chirimen-drivers/CONTRIBUTING#%E6%8E%A8%E5%A5%A8%E3%81%99%E3%82%8B%E3%83%A2%E3%82%B8%E3%83%A5%E3%83%BC%E3%83%AB%E3%81%AE%E6%A7%8B%E6%88%90) として言及しているだけで実際にどうすれば良いのかという説明が無い。

一般の npm モジュールなどのリポジトリとしてはそれでいいのだけが、既存の形式から切り替えるに際して、今後はこのように開発手順を変更してください・このように書いてくださいという ESM への以降に関する説明・合意が不十分だった。

故に、推奨構成のモジュールの書き方、旧来の `<scrpt src=xxx.js>` から `<script src=xxx.mjs type=module>` に変える時の手順について丁寧に書いた案内ドキュメントを追加したい。

dynamis, 2020-12-10:
想定内容:
- [ ] js は mjs から自動生成する旨の説明
- [ ] js/mjs それぞれの読み込み方を説明
- [ ] 既存 js ドライバの ｍjs 化の手順説明
  - [ ] mjs にして export を追加
  - [ ] テストコードの html,main.js に script タグと import の変更
- [ ] リンク(リファレンス)
  - https://developer.mozilla.org/ja/docs/Web/JavaScript/Guide/Modules

satakagi, 2020-12-10:
https://github.com/chirimen-oh/chirimen-drivers/issues/94#issuecomment-742267309
にコメントしました。　こっちにコメントしたほうが良かったかもですね。

dynamis, 2020-12-10:
方針の議論なので #94 の方で良いです。

移行プランにかかわらず必要だろう、esm の書き方の説明ドキュメントを書く部分だけ (他の議論と一緒流れないよう) 分離しただけなので。

kou029w, 2020-12-12:
パッケージを作るための手順を説明したパッケージを作りました https://www.npmjs.com/package/@chirimen/hello-world

satakagi, 2020-12-19:
yarnというのをnpmでinstallしてyarn && yarn build
してもERRORが出て止まります。
yarnの役割を理解していないのでトラブルシュートできません。

yarnなどを使わない、もっと簡単なリリースの手順にしていただけないでしょうか？

kou029w, 2020-12-19:
Yarnとはパッケージ管理のためのNpmと似た役割のアプリケーションです。
目的としては一度に複数のパッケージをモノレポの中で扱うためです。
Npm v7 より前のバージョンではworkspacesプロパティをサポートしておらずこのリポジトリを作る時に不便だったのでYarnを採用しました。
yarn buildでエラーがあるということは同様にnpm run buildでもエラーがある、ということで、それはそれとして対処したほうが良いですが、それは一旦置いておいて、Npm v7 以降であるならworkspacesをサポートしてるのでYarnをやめることは可能で、代わりにv7以降の npm コマンドを使うだけですね。

kou029w, 2020-12-19:
- 自動的にPublishされるように設定しておきました ([GitHub Actions](https://github.com/chirimen-oh/chirimen-drivers/blob/b5176f83d1e1f0a9f592b3c4e78d9e5c6063941d/.github/workflows/release.yml))
- [リリース方法も手動でnpm publishせず済むようになった](https://chirimen.org/chirimen-drivers/CONTRIBUTING#%E3%83%AA%E3%83%AA%E3%83%BC%E3%82%B9%E6%96%B9%E6%B3%95)ため簡単になりました