chirimen-oh/chirimen-drivers #149: node.jsでESModuleとして使おうとするとエラーアウトする

satakagi, 2021-08-31:
> To load an ES module, set "type": "module" in the package.json or use the .mjs extension.

各ドライバのpackage.jsonに、"type": "module"を追加することで動作するようになりました。
が、サイドエフェクトは無いでしょうか？

kou029w, 2021-08-31:
副作用はあると考えます。CHIRIMEN環境でバンドラー等を介して従来の index.js を使おうとすると意図せず読み込めない可能性が考えられます。
ところで、下記ではインポートに失敗しますか?

```js
import ADT7410 from "@chirimen/adt7410";
```

参考: README.md

satakagi, 2021-08-31:
あ、READMEに書かれていたんですね。
ちなにみこの書き方でうまくライブラリが読み込めるのは何かnode.js特有の機能に依るのでしょうか？
ファイルパスを明示しなくても読み込めているので何か特別な機能なのかなと思いますが。

kou029w, 2021-08-31:
> ちなにみこの書き方でうまくライブラリが読み込めるのは何かnode.js特有の機能に依るのでしょうか？

はい。おっしゃるとおりです。こちらのドキュメントにあるNode.jsの振る舞いです。
ドキュメント: https://nodejs.org/api/packages.html

kou029w, 2021-08-31:
特有かどうか、という観点で言うと、実は特有ではないですね。訂正します。
標準化の提案はTC39 [tc39/proposal\-import\-assertions: Proposal for syntax to import ES modules with assertions](https://github.com/tc39/proposal-import-assertions) にてあり、現時点ではStage 3です。

satakagi, 2021-08-31:
https://github.com/chirimen-oh/chirimen.org/tree/master/pizero/es-examples/ADT7410
のサンプルは
```javascript
import ADT7410 from "@chirimen/adt7410";
```
でつくりました。closeしますね。

closed 2021-08-31 by satakagi