chirimen-oh/chirimen-drivers #403: ドライバのコントリビューション方法の説明が不十分

satakagi, 2025-08-26:
[https://chirimen.org/chirimen-drivers/CONTRIBUTING](https://chirimen.org/chirimen-drivers/CONTRIBUTING
のドキュメントの拡充が必要と思われます。

kou029w, 2025-12-27:
https://github.com/chirimen-oh/chirimen-drivers/pull/444 にて説明拡充しました！

> # Contributing Guidelines
> 
> [CHIRIMEN Drivers](https://github.com/chirimen-oh/chirimen-drivers) へのコントリビュート方法です。
> 
> ## 目次
> 
> - [行動規範](#行動規範)
> - [Issues と Pull Request](#issues-と-pull-request)
> - [プロジェクト構成](#プロジェクト構成)
> - [開発環境のセットアップ](#開発環境のセットアップ)
> - [初めての貢献：ドキュメント修正](#初めての貢献ドキュメント修正)
> - [新しいドライバの追加：完全ガイド](#新しいドライバの追加完全ガイド)
> - [既存ドライバの修正方法](#既存ドライバの修正方法)
> - [リリース方法](#リリース方法)
> - [よくある質問](#よくある質問)
> 
> ## 行動規範
> 
> [コントリビューター行動規範 - Contributor Covenant](https://www.contributor-covenant.org/ja/version/1/4/code-of-conduct)
> 
> 行動規範に反するコントリビュートは受け入れません。
> 
> ## Issues と Pull Request
> 
> ### Issues
> 
> 質問、バグ、提案、その他 Issue を歓迎します。
> 
> - **バグ報告**: 再現手順、期待する動作、実際の動作を記載してください
> - **機能提案**: 具体的なユースケースと共に提案してください
> - **質問**: 遠慮なくお尋ねください
> 
> ### Pull Request
> 
> ドキュメント追加、モジュール追加、バグ修正、その他改善するための Pull Request を歓迎します。
> 
> ## プロジェクト構成
> 
> このリポジトリは [npm workspaces](https://docs.npmjs.com/cli/v8/using-npm/workspaces) と [Lerna](https://lerna.js.org/) で管理される **monorepo** です。複数の CHIRIMEN ドライバパッケージを一つのリポジトリで管理しています。
> 
> ### ファイル構造
> 
> ```
> /
> ├── LICENSE
> ├── CONTRIBUTING.md # このドキュメント
> ├── README.md # プロジェクトのトップページ
> ├── .github/workflows/ # GitHub Actions ワークフロー
> │ ├── ci.yml # CI (npm ci の実行)
> │ └── release.yml # リリース自動化
> ├── examples/ # ドキュメント用のMarkdownファイル
> ├── raspi-examples/ # CHIRIMEN for Raspberry Pi ブラウザ用サンプルコード
> ├── microbit-examples/ # CHIRIMEN with micro:bit 用サンプルコード
> ├── node-examples/ # Node.js 用サンプルコード
> │ └── adt7410/ # 各ドライバのサンプル
> │ ├── main.js
> │ └── package.json
> ├── lerna.json # Lerna 設定ファイル
> ├── package.json # ルート package.json (workspaces 設定)
> └── packages/ # ドライバモジュールのソースコード
> ├── hello-world/ # サンプルドライバ
> ├── adt7410/ # 温度センサー (ADT7410)
> │ ├── index.js # ドライバ本体
> │ ├── package.json # パッケージ設定
> │ └── README.md # ドライバのドキュメント
> ├── amg8833/ # 赤外線アレイセンサー
> └── ... # その他60個以上のドライバ
> ```
[... 485 lines cut ...]
> **A.** 以下のドライバが参考になります:
> 
> - **シンプルな例**: `packages/hello-world/` - 最小構成
> - **実用的な例**: `packages/adt7410/` - 温度センサー
> - **複雑な例**: `packages/amg8833/` - 赤外線アレイセンサー
> 
> ### Q5. I2C 以外のデバイス（GPIO、SPI など）のドライバも追加できますか
> 
> **A.** はい、可能です。このリポジトリは主に I2C デバイスを扱っていますが、GPIO や SPI を使用するドライバも歓迎します。
> 
> ### Q6. ドキュメントは日本語で書くべきですか
> 
> **A.** README は日本語で記述することを推奨します。コード内のコメントやコミットメッセージは英語でも日本語でも構いません。
> 
> ## 質問・相談
> 
> 不明点があれば、お気軽に [Issue](https://github.com/chirimen-oh/chirimen-drivers/issues) を作成してください。
> 
> ## 参考リンク
> 
> - [CHIRIMEN 公式サイト](https://chirimen.org/)
> - [npm workspaces ドキュメント](https://docs.npmjs.com/cli/v8/using-npm/workspaces)
> - [Lerna ドキュメント](https://lerna.js.org/)
> - [セマンティックバージョニング](https://semver.org/lang/ja/)
>

gurezo, 2026-02-21:
@satakagi @tadfmac @kou029w @sizuhiko 

ChatGPT, Gemini に [chirimen-oh.plan.md](https://github.com/user-attachments/files/25457337/chirimen-oh.plan.md) を添付して、解決案を出させてみました。

chirimen-drivers の**理想像**は、こんなイメージでどうでしょうか？（移行手順がさておき

```
chirimen-drivers/
├── .github/
│   └── CONTRIBUTING.md          # 共通のコントリビュート手順（新規追加・修正ルール）
├── devices/                     # デバイス情報の一元管理エリア
│   ├── partslist.csv            # 全デバイスのメタデータ（旧 chirimen.org より移行）
│   └── images/                  # デバイスの画像データ（旧 partsImgs より移行）
├── drivers/                     # I2C / GPIO ドライバの実装本体
│   ├── i2c-ads1015.js
│   └── ...
├── examples/                    # 実行環境ごとのサンプルコード集約
│   ├── esm/                     # 最新のWeb標準(ESM)サンプル（旧 pizero/esm-examples）
│   │   ├── index.html           # index_examples.csv から生成されるポータル
│   │   └── [device-id]/
│   │       └── main.js
│   ├── node/                    # Node.js環境用（node-web-gpio/i2c利用）
│   ├── raspi/                   # Raspberry Pi 固有の実装（旧 chirimen リポジトリより）
│   └── microbit/                # Web Bluetooth用（旧 chirimen-micro-bit より）
├── tools/                       # メンテナンス用スクリプト
│   └── generate-index.js        # CSVから index.html を自動生成するツール
└── README.md                    # リポジトリ全体のガイド
```